360Controller/360Controller #1162: XBox One Wireless Controller + USB-C 'No Devices Found' on Mac Mojave 10.14.6

dsantiqueza, 2021-05-10:
### Type of Controller
XBox One Wireless Controller + USB-C

### OS Version
macOS Mojave Version 10.14.6

### Driver Version
Xbox 360 Controller Driver Version 1.0.0-alpha.6

### Connection Method
Wired / Bluetooth

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x0b12
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Vibrates when initially connected, but does not light up unless pressed.

### Response in Preference Pane
In System Preferences, under the 'Xbox 360 Controllers', it says 'No devices found' in the dropdown menu.

### Detailed Explanation of Issue
No matter what I do, it just says 'No devices found' on the Xbox 360 Controllers section of the System Preferences Pane.

I have installed, uninstalled, and reinstalled the Driver several times. I have also reconnected the Controller several times. I have primarily followed [this thread](https://github.com/360Controller/360Controller/issues/911) to troubleshoot my issue, but unfortunately it isn't working and I'm still at a loss for what to do.

For reference, these are the results of the Terminal commands:

1. sudo kextutil /Library/Extensions/360Controller.kext
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

2. sudo kextcache --clear-staging
3. sudo kextutil /Library/Extensions/360Controller.kext
Error copying bundle during staging: -1
Unable to stage kext (/System/Library/Extensions/roller.kext) to secure location.
Notice: /Library/Extensions/360Controller.kext has debug properties set.
Warnings: 
    Personality CFBundleIdentifier differs from containing kext's (not necessarily a mistake, but rarely done): 
        ChatPadKeyboardEvents

I initially had not allowed permissions to "Drew Mills", but I have since allowed it and now even typing the 'sudo kextutil /Library/Extensions/360Controller.kext' I do not get another prompt asking for approval.

I apologize for the trouble, but I would appreciate any help in resolving this issue.

Thank you very much!

dsantiqueza, 2021-05-10:
My apologies, my brother just told me that actually what I have is an 'Xbox Series X|S Controller'. I'm not sure if this changes anything.

Thank you.

Scarsz, 2021-09-17:
Same issue with same controller on Catalina. 4th generation controller model 1914.

benoit-girard, 2021-12-02:
Same issue with the same controller and the same macOS version.
No hint of what can be done?