360Controller/360Controller #1163: Request to add new controller Nacon GC-100XF, Vendor ID: 0x3285 / Product ID: 0x0607

jsimon, 2021-05-15:
### Type of Controller
3rd Party

### OS Version
10.14.6 (Mojave)

### Driver Version
360Controller 1.0.0-alpha.6
360Controller 1.0.0-alpha.2

### Connection Method
Wired

### Device Name and Info
Device Name: Nacon GC-100XF
<br/>Product ID: 0x0607
<br/>Vendor ID: 0x3285

The product ID is in XInput mode. Product ID in DirectInput mode is 0x0c03.
There is already an entry for Nacon GC-100XF in the Info.plist, but with a different product and vendor ID. 

### Response of Controller
No lights (In DirectInput mode all lights light up)

### Response in Preference Pane
Not recognized

### Detailed Explaination of Issue
The controller is not recognized. This is no surprise as the vendor/product IDs are not mentioned in the Info.plist.
As there seem to be people who have successfully used this controller (with the existing vendor/product) I hope that it could work by adding the IDs.

g0ara, 2021-12-03:
Hello,

I've got the same controller with the same IDs. I tried to follow the instructions from the readme (Adding Third Party Controllers) but that is clearly beyond my abilities.
Is there any chance someone could add the IDs to the Info.plist?
Thanks!

toshe, 2022-04-08:
I added the XInput Vendor ID and Product ID to Info.plist. Let's hope somebody can compile and build with this old targets.