360Controller/360Controller #1165: Add Device: "Afterglow Prismatic Wired Controller for Xbox One"

idolpx, 2021-05-25:
### Type of Controller
Xbox One

### OS Version
macOS Big Sur 11.4

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Afterglow Prismatic Wired Controller for Xbox One
<br/>Product ID: 0x0139
<br/>Vendor ID: 0x0e6f

### Response of Controller
No devices found

### Response in Preference Pane
No devices found

### Detailed Explaination of Issue
I'd like for this device to be added to the list of supported controllers.  Thanks!

Sofie-Dragon, 2021-05-26:
I second this!! I have the same controller and just cracked into my school computer! (Dw I'm not in trouble ;)

thebeardedcrazycatlady, 2021-06-02:
I recently got the same controller.
I am unable to uninstall or update this driver, it always fails.
I get this error when attempting to enable the driver:
" The driver has failed to load due to an unknown error, code: -603947007. "