360Controller/360Controller #1168: Request to add new controller SHANWAN  Vendor ID: 0x045e / Product ID: 0x028e

baldscone, 2021-06-20:
### Type of Controller
3rd party xBox https://www.amazon.co.uk/gp/product/B0893TGQPL/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### OS Version
10.14.6 (Mojave)

### Driver Version
360Controller 1.0.0-alpha.6

### Connection Method
Wired

### Device Name and Info
Device Name: SHANWAN
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e

### Response of Controller
Blue power light on

### Response in Preference Pane
1: Xbox 360 Wired Controller (Wired) appears in the preference pane but no buttons or joysticks respond.

### Detailed Explaination of Issue
The controller appears in preference pane but no inputs work at all. Below information from System report. Would be great if someone could add it or help me trouble shoot. Thank you.

Controller:

  Product ID:	0x028e
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.10
  Speed:	Up to 12Mb/sec
  Manufacturer:	SHANWAN
  Location ID:	0x14200000 / 6
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0

s4shubham4u, 2021-07-23:
Well, if u change the mode to gamepad the controller will probably work no vibration thou

toshe, 2022-04-08:
Can you please add a new entry to Info.plist and submit a pull request?

zellwk, 2023-08-25:
@toshe how do I navigate to `360Controller/360Controller/Info.plist`? Mind sharing the full path to this file?

AndrewPrzelucki, 2024-01-08:
Rumble support for 2563:0526 please? Named "SHANWAN Android Gamepad", it's a wireless USB dongle pad. Currently handled by usbhid - all axes and buttons working, but no rumble! I managed to get it working for a session with a script for another model, but I can't do it again. Under usbhid it's locked in analog mode as well. [Here's the store page and Windows driver (link).](https://esperanza.pl/esperanza-gamepad-bezprzewodowy-2-4ghz-ps3-pc-usb-gladiator-czarny,176,1697.html)