360Controller/360Controller #1169: Driver shows up in system preferences but isn't detecting controller

biggiepee, 2021-07-02:
### Type of Controller
Replace this text with: Xbox One

### OS Version
Version 11.4, Big Sur

### Driver Version
360Controller 1.0.0-alpha.6

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller
<br/>Product ID: 0x02d1
<br/>Vendor ID: 0x045e

### Response of Controller
Guide button glows and pulses but Never actually stays on, controller remains undetected and unresponsive

### Response in Preference Pane
Controller is not detected

### Detailed Explaination of Issue
I connected my xbox controller to my MacBook, but when I installed the driver, it gave me some kind of error and said that I needed to contact the developer. I opened system preferences anyway and checked and saw that the driver was there anyway. I tried turning on my controller to no avail. I went to advanced and clicked on enable this driver and the whole system preferences window froze. When I click uninstall this driver in advanced, it gives me an error "Error Uninstalling the Driver!"

vasanthtt, 2021-07-04:
I get same issue

nash8114, 2021-08-02:
Please refer to https://github.com/360Controller/360Controller/issues/1139