360Controller/360Controller #1171: CARBON BLACK Xbox Controller not found

BernardFaucher, 2021-07-10:
### Type of Controller
Xbox Series X|S, Xbox One

### OS Version
MacOS Version 10.14.6 (18G9216)

### Driver Version
360Controller 1.0.0-alpha.6

### Connection Methods
Wired: USB-C to USB-C
Wired: USB-C to USB-A,  with adapter: USB-A to USB-C

### Device Name and Info from `Hardware -> USB`
```
  Controller:

  Product ID:	0x0b12
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	5.01
  Serial Number:	3039373131393336373836303236
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft
  Location ID:	0x14600000 / 6
  Current Available (mA):	500
  Extra Operating Current (mA):	0
```

### Response of Controller
* Controller lights up when power button pressed.
* Vibration on wired connection.

### Response in Preference Pane
Controller does not appear in the preference pane - "No devices found".

### Detailed Explaination of Issue
Xbox controller "CARBON BLACK" is visible to MacOS over wired connection, but not visible to XBOX 360 Controllers preference pane. Similar (same?) issue reported in #1162 and perhaps others. Thanks!

toshe, 2022-04-08:
Can you please add a new entry to Info.plist and submit a pull request?

stevenjamescox, 2022-05-24:
This should controller will be added by pull request #1224.