360Controller/360Controller #1179: Uninstalling from Big Sur impossible

foxbro8, 2021-08-02:
### Type of Controller
N/A

### OS Version
11.5.1 Big Sur

### Driver Version
1.0.0-alpha.6

### Connection Method
N/A

### Device Name and Info
N/A

### Response of Controller
N/A

### Response in Preference Pane
N/A

### Detailed Explanation of Issue
Unfortunately, the only error message I am given is "Error! Error uninstalling the driver!" When I used the manual code it said uninstallation was successful, but every file still was on my computer and it was still on my computer. Even after manually resetting my computer, no solution was created.

suxiaogang, 2021-08-02:
u could have a try with software like CleanMyMacX. find it in system preferences panel, check and click remove

foxbro8, 2021-08-02:
Tried that out and it didn’t work. It’s not even showing up in the CleanMyMac scan. It’s as if the app doesn’t exist because it uninstalled, except it doesn’t uninstall at all.

LTHoriZonT, 2021-08-05:
Did you try reinstalling 360driver and using CMC to uninstall?

foxbro8, 2021-08-06:
Yes so after it was all messed up I decided to re install it to have my system run a check on it and every time I remove it, it leaves it on the computer and leaves files I can’t move, never mind not uninstalling at all.

sriramr98, 2021-08-09:
+1. Need a fix for this please

hcyang22, 2021-08-13:
I am having the same problem, and CleanMyMacX didn't help.

tematkchv, 2021-08-15:
same problem! anyone know where it may be located in library?

fullife32, 2021-09-03:
Open Terminal.app and type this command "cd ~/Library/PreferencePanes && open .", it will open the place where the driver is installed, you just have to delete it and restart your mac !

mikeabrantes, 2021-10-19:
Using CleanMyMac you will find it to remove. Follow the steps:
Open CleanMyMac > Go to Extensions > View All Extensions... > Preference Panes > Select the 360controller driver and click remove. Done.

m1nkeh, 2021-12-04:
> Using CleanMyMac you will find it to remove. Follow the steps: Open CleanMyMac > Go to Extensions > View All Extensions... > Preference Panes > Select the 360controller driver and click remove. Done.
> 
> If you dont have CleanMyMac paid version do this: Install an older 360controller version on top of the one you have it. I suggest installing the v2. Now go to the driver > Advanced > Unistall... Done

can confirm, this does indeed work. Ty!