360Controller/360Controller #1180: a simple approach to make it work on Big Sur

suxiaogang, 2021-08-03:
hi guys, I just find a simple way to make it work on Big Sur (Intel chip ONLY !) 

1. download it at https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.6
2. open the dmg file and drap `Install360Controller.pkg` to your desktop
3. on your command line
``` 
cd ~/Downloads
pkgutil --expand ./Install360Controller.pkg ./360Controller
``` 

use a texteditor to open it (I use VS code)
code ./360Controller/Install360Controller.pkg/Scripts/postinstall

```
#!/bin/sh

# Get the major and minor versions of the current OS
MAJOR_VER=$(sw_vers -productVersion | awk -F. '{ print $1; }')
MINOR_VER=$(sw_vers -productVersion | awk -F. '{ print $2; }')

# Just have to make sure that macoS didn't update to 11 and really mess things up
if [[ "${MAJOR_VER}" -ne 10 ]]; then
    echo "Major version is not 10. Cowardly failing out of script execution."
    exit 1
fi

# This is to fix a bug in Yosemite.
if [[ "${MINOR_VER}" -le 14 && "${MINOR_VER}" -ge 10 ]]; then
    /bin/ln -s /Library/Extensions/360Controller.kext /System/Library/Extensions/
fi

# Make the uninstall script in the preference pane executable
/bin/chmod ug+x /Library/PreferencePanes/Pref360Control.prefPane/Contents/Resources/upgrade.sh

/usr/sbin/kextcache -i /
/usr/bin/touch /Library/Extensions

/bin/launchctl load -w /Library/LaunchDaemons/com.mice.360Daemon.plist

exit 0
```

comment the first 4 lines which are:

```
# if [[ "${MAJOR_VER}" -ne 10 ]]; then
#     echo "Major version is not 10. Cowardly failing out of script execution."
#     exit 1
# fi
```
4. repack
```
pkgutil --flatten ./360Controller Install360Controller_new.pkg 
```
5. install Install360Controller_new.pkg , done.

pmele08, 2021-08-10:
can you attach the completed file I can't seem to get it to work. Thanks!

flyingfinn64, 2021-08-16:
The solution while bypasses the installation problem still doesn't make driver work. 
Enabling driver leads to an unknown error. 
Cant be enabled. Though it can be uninstalled through the preferences panel.

grosser, 2021-11-06:
same, install works but then get error :(
[image]

letalvoj, 2021-12-04:
I managed to make this work by 

1. installing `0.16.11` (successfully, yet the gamepad is not discovered)
2. installing `1.0.0-alpha6`, which fails - yet it overwrites some files anyway, which fixes the issue of `0.16.11`

t151848p, 2022-03-31:
> I managed to make this work by
> 
>     1. installing `0.16.11` (successfully, yet the gamepad is not discovered)
> 
>     2. installing `1.0.0-alpha6`, which fails - yet it overwrites some files anyway, which fixes the issue of `0.16.11`

I'm able to install the driver with this (it shows up under System Preferences) but trying to click "Enable Driver" in the Advanced tab still runs into the same error.

```
The driver has failed to load due to an unknown error, code: -603947007. Please report this error to the driver maintainer.
```