360Controller/360Controller #1184: The driver has failed to load due to an unknown error, code: -603947007. Please report this error to the driver maintainer.

passos, 2021-08-28:
### Type of Controller
Xbox One

### OS Version
Big Sur 11.5.2

### Driver Version
1.0.0-alpha.6

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
>Device Name: Controller
<br/>Product ID: 0x02dd
<br/>Vendor ID: 0x045e  (Microsoft Corporation)
```
  Product ID:	0x02dd
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	2.03
  Speed:	Up to 12 Mb/s
  Manufacturer:	Microsoft
  Location ID:	0x00231000 / 7
  Current Available (mA):	500
  Extra Operating Current (mA):	0
```

### Response of Controller
Connect controller to MacBook Pro with USB cable. The controller was vibrated when cable connected. 

### Response in Preference Pane
The controller does not appear in the preference pane

### Detailed Explaination of Issue
The controller is connected to MacBook Pro with an USB cable. The controller was vibrated when cable connected. 

But it shows "No device found" in the preference pane "Xbox 360 Controller".

I switch to "Advanced" tab and checked the checkbox "Enable Driver". It shows following message after few seconds:
```
The driver has failed to load due to an unknown error, code: -603947007. Please report this error to the driver maintainer.
```.

DJTechZombie, 2021-11-10:
I am having the same issue. I tried a "fix" from #1186 "Install Problems" without any luck.

PxBCPG, 2022-03-12:
Same problem on macOS Big Sur (intel). 
Any clues?

lucasrowe, 2022-06-04:
Getting this on Monterey 12.3.1

khaleds-brain, 2022-06-16:
> Getting this on Monterey 12.3.1

Same here

weicong96, 2022-06-21:
same here Monterey 12.0.1

gabrielucido, 2022-11-13:
Same on MacOs Ventura 13.0.1

waldo68, 2023-01-02:
me too
product 0x581b
vendor 0x24c6

Infixo, 2023-07-26:
Exactly same issue and error code.
However, it worked when I installed it on 12.04.2023. Now I tried to play again and it works no more.