360Controller/360Controller #1185: It doesn't let me uninstall it

CrazyCat4284, 2021-08-29:
### Type of Controller
N/A

### OS Version
11.5.2

### Driver Version
?

### Connection Method
?

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:MacBook Air
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

fullife32, 2021-09-03:
Open Terminal.app and type this command "cd ~/Library/PreferencePanes && open .", it will open the place where the driver is installed, you just have to delete it and restart your mac !

fatevase, 2022-01-29:
if u want remove prefPane in setting, 
u need execute `uninstall.commands`(download from assets) first.
then open `/Library/PreferencePanes` and delete `Pref360Control.prefPane`,
done.