360Controller/360Controller #1186: Install Problems

zmaneman, 2021-09-09:
### Type of Controller
 N/A

### OS Version
Catalina 11.4

### Driver Version
Alpha 6

>Device Name: MacBook Pro 2017

### Detailed Explaination of Issue
I am getting an error when attempting to install the driver. I manually uninstalled after updating (Downgrading) to Catalina from Mojave when my controller wasn't working. Attempted to get the latest version of the driver but nothing will work. Pic shows the message I get when I try to install. It will show up in the preferences pane, but wired controller will blink twice when connected then shut off, and the driver window says no devices connected. Help please! Is there any way at all to make a wired only 360 controller work with Catalina?
[image]

neils2, 2021-09-13:
Having the same issue as above in the screenshot

Aliciyar, 2021-09-14:
Having the same issue but on Big Sur 11.5.2/M1 chip 
says the developer needs to update 

[image]

benchman, 2021-09-16:
Found a way to make this work on Big Sur.
You basically need to follow this approach [,](https://steamcommunity.com/discussions/forum/2/3004429475624592660/) but just change the signing team id of the kext with Drew's one.

Replace this:
spctl kext-consent add MXGJJ98X76
with this
spctl kext-consent add G947KKM5RL

ljstary, 2021-10-30:
I have the same issue with Alpha 6 failing to install on an Intel MacBook Pro running Big Sur. 

OS: Big Sur 11.6.1 (20G224)
Hardware: MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports), 2.3 GHz Quad-Core Intel Core i5

I had previously installed 0.16.11 but it wasn't detecting my wired Xbox 360 controller. 

I haven't tried benchman's workaround.

DJTechZombie, 2021-11-10:
I tried the workaround, and I still can't get it to recognize my controller (rock candy wired 360 controller. 

When I click "Enable Driver", it says "The driver has failed to load due to an unknown error, code: 603947007"

mcsantiago, 2021-11-30:
Can report that this problem exists on M1 macbooks as well.

macOS Monterey

biller-aivy, 2022-10-16:
Alpha 6
[image]
MacOS Ventura - M1