360Controller/360Controller #1187: Product ID: 0x028e

avalanche1, 2021-09-12:
### Type of Controller
Xbox 360

### OS Version
10.14

### Driver Version
1.0.0-alpha.6

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: 
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e

### Response of Controller
Absolutely dead. Central button flashes for a moment, when plugging into the usb port.

### Response in Preference Pane
The controller is not detected on xbox 360 controllers pref pane.

### Detailed Explaination of Issue
Absolutely dead. Not detected neiter by Steam, nor absolutely any game.

toshe, 2022-04-08:
Can you please add a new entry to Info.plist and submit a pull request?

stevenjamescox, 2022-05-18:
@avalanche1 For what it's worth, I have that exact same controller and had the same trouble trying to get it working on 10.14 using **1.0.0-alpha.6**.
Fortunately, though, I did find success with **1.0.0-alpha.5**.  Perhaps give Alpha 5 a whirl for 10.14.