360Controller/360Controller #1194: Xbox Controller not showing

joektate, 2021-11-10:
### Type of Controller
Xbox One, Wireless not Bluetooth

### OS Version
Big Sur

### Driver Version
Newest

### Connection Method
Wired through USB

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:0x02dd
<br/>Vendor ID:0x045e

### Response of Controller
No response

### Response in Preference Pane
Not even detecting it, tried all different versions of the driver from 2 upward

### Detailed Explaination of Issue
The controller vibrates when plugged in but shows nothing, looked into the hardware and it does show there but not in the driver through system preferences, even when I press the xbox button, it doesn't show anything

sean-shore, 2021-12-12:
I had the same issue and believe it is due to the driver not being allowed to run. Try this as it has worked for me:

- Boot into Recovery mode (hold ⌘-R at boot)
- Open Terminal (Utilities -> Terminal).
- Type
    `spctl kext-consent add G947KKM5RL`
- and press return.
- Type
    `kmutil trigger-panic-medic --volume-root "/Volumes/Macintosh HD"`
- (replace “Macintosh HD” with whatever your drive is named; press Tab at the end of typing "/Volumes/" to get a hint).
- Reboot.

When you restart a popup will appear asking you to allow the developer 'Drew Mills', after you allow this developer restart again and the controller should work.

andrevsm, 2022-01-03:
> I had the same issue and believe it is due to the driver not being allowed to run. Try this as it has worked for me:
>
> * Boot into Recovery mode (hold ⌘-R at boot)
> * Open Terminal (Utilities -> [...]

I tried but it does not work.. Im trying with Xbox One Controller :(

0uxh, 2022-01-08:
> 

Hey @sean-shore 

Attempted what you posted here, but it responded with `spctl: command not found` when I tried this in recovery mode. Out of curiosity I checked what would happen if I just did it from a terminal window from within macos and it seemed to recognize the command but not allow me to execute the command from there (response was `spctl: failed to store new configuration.`)

Would you happen to know what's going wrong here? I'm fairly close to clueless with this stuff, aside from following basic instructions on github 😓