360Controller/360Controller #1199: 1.0.0-alpha 6 won'

syntheticperson, 2021-12-10:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

sean-shore, 2021-12-12:
I had the same issue and this fixed it for me:

-     Boot into Recovery mode (hold ⌘-R at boot).
- Open Terminal (Utilities -> Terminal).
- Type
    `spctl kext-consent add G947KKM5RL`
- and press return.
- Type
    `kmutil trigger-panic-medic --volume-root "/Volumes/Macintosh HD"`
- (replace “Macintosh HD” with whatever your drive is named; press Tab at the end of typing "/Volumes/" to get a hint).
- Reboot.

When you restart a popup will appear asking you to allow the developer 'Drew Mills', after you allow this developer restart again and the controller should work.

PsyPryss, 2022-01-11:
Installation does fail. It will still ask for permission in your Security and Privacy tab. It won't detect any controllers in the new Xbox 360 controller tab. But dang it it still works when you start the game.

Alexaandre, 2025-02-23:
> I had the same issue and this fixed it for me:
>
> * Boot into Recovery mode (hold ⌘-R at boot)
> * Open Terminal (Utilities -> Terminal).
> * Type
>   `spctl kext-consent add G947KKM5RL`
> * and press return. [...]

It worked for me ! After reboot it was already checked in grey but it says on screen drew mills "approved - updated" i had to reinstall again then reboot then controller was recognized