360Controller/360Controller #1200: Error code-603947007.

WalteerDev, 2021-12-17:
### Type of Controller
Replace this text with: All

### OS Version
macOS Monterey .1 (new recent update)

### Driver Version
alpha 6

### Connection Method
Replace this text with: Wired

### Device Name and Info
we don't care cause I didn't connected the device, I just tried to enable the driver and it failed

### Response of Controller
"The driver has failed to load due to an unknown error, code: -603947007. Please report this error to the driver maintainer."

### Response in Preference Pane
I just installed and this error...

oliviapriefert, 2021-12-20:
same issue... also not recognizing controller

dwaam, 2021-12-21:
Same issue with Macbook pro M1 with Monterey.

WalteerDev, 2021-12-24:
On my macbook pro 14 m1pro 2021 nothing to do, it doesn't want to install

Infixo, 2023-07-26:
Damn, I have the same issue. Came here hoping for a solution... :( :(

TheGamesss, 2023-11-08:
Same issue with my M1 Mac mini.