360Controller/360Controller #1201: No devices found on macOS Monterey with PDP Afterglow Controller

aadictive, 2021-12-17:
### Type of Controller
PDP Afterglow Wired Controller

### OS Version
macOS Monterey v 12.1

### Driver Version
Tried:
360Controller 1.0.0-alpha.6
360Controller 1.0.0-alpha.5
360Controller 1.0.0-alpha.4
360Controller 0.16.11

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Afterglow Wired Controller for Xbox Series X:
<br/>Product ID: 0x02da
<br/>Vendor ID: 0x0e6f

### Response of Controller
Only the backlit lights glow, not the main connection led.

### Response in Preference Pane
N/A

### Detailed Explanation of Issue
The installation is successful after trying the methods in previous issues. eg: https://github.com/360Controller/360Controller/issues/1199#issuecomment-991886466
But No Devices Found message in the System Settings 360Controller Pane. I don't know but it simply cannot recognize or the drivers are not compatible.

I guess the drivers are not yet available for updated MacOS, they work on windows well. Kindly suggest!

[image]
[image]

Ibrokethebuild, 2021-12-25:
I have the exact same issue.
A fix would be most welcome.

bagelface, 2021-12-26:
Also looking for a fix for this

cssagogo, 2022-01-04:
I'm having same problem on Monterey since upgrading with a wired xbox controller knock off that worked fine prior to updating OS.

marcuschristiansen, 2022-04-01:
Same issue here.

toshe, 2022-04-08:
Can you check if this device is in Info.plist and if not add it device to Info.plist as a new entry and create a Pull Request ?