360Controller/360Controller #1202: The installation failed, no devices found.

Piipperi, 2021-12-27:
### Type of Controller
Xbox One S Controller

### OS Version
12.1

### Driver Version
1.0.0 Alpha 6

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: N/A
<br/>Product ID: N/A
<br/>Vendor ID: N/A

### Response of Controller
It vibrates when connected to Mac but nothing else happens

### Response in Preference Pane
No device found

### Detailed Explaination of Issue
Installer fails to install, but I did allow the kext to load and I have rebooted the machine and tried installing the driver afterwards. The preference panel works.

BrutchsamaJeanLouis, 2022-01-07:
Same here. It said install failed but afterwards i still get the system preference menu. which doesn't respond or detect a device.

McPaddy85, 2022-01-09:
Also same!

MattDemers, 2022-01-17:
Doesn't uninstall easy, either.