360Controller/360Controller #1205: Impossible to uninstall 360controller

mihk3, 2022-01-30:
### Type of Controller
Replace this text with: N/A

### OS Version
MacOS Monterey 12.0.1

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Replace this text with: N/A

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: N/A
<br/>Vendor ID: N/A

### Response of Controller
N/A

### Response in Preference Pane
[image]

### Detailed Explaination of Issue
I'm unable to uninstall 360controller and all the drivers from my mac. I've tried almost everything but nothing works. Really worried right now. For security reasons I want this gone from my mac as soon as possible. Thanks in advance!

mihk3, 2022-01-30:
I was also unable to quit "360Daemon" from the activity monitor.

JC-333001, 2022-02-17:
I have the same problem. Does anyone know how to fix this?

stanjg, 2022-02-18:
+1 Having the same problem, none of my controllers are being detected either.

I'm on macOS Montery 12.1, with an M1 macbook pro.

gutasiadam, 2022-03-06:
Same problem here. Does anyone have a solution?

zhong015, 2022-03-14:
do shell script "
        
launchctl unload /Library/LaunchDaemons/com.mice.360Daemon.plist
kextunload -b "com.mice.driver.Xbox360Controller"
kextunload -b "com.mice.driver.Wireless360Controller"
kextunload -b "com.mice.driver.WirelessGamingReceiver"
rm -f  /Library/LaunchDaemons/com.mice.360Daemon.plist
rm -rf /Library/Application\ Support/MICE/360Daemon.app
rm -rf /System/Library/Extensions/360Controller.kext
rm -rf /System/Library/Extensions/Wireless360Controller.kext
rm -rf /System/Library/Extensions/WirelessGamingReceiver.kext
rm -rf /Library/Extensions/360Controller.kext
rm -rf /Library/Extensions/Wireless360Controller.kext
rm -rf /Library/Extensions/WirelessGamingReceiver.kext
rm -rf /Library/Extensions/XboxOneBluetooth.kext
rm -rf /Library/PreferencePanes/Pref360Control.prefPane
pkgutil --forget com.mice.pkg.Xbox360controller

        " with administrator privileges

gutasiadam, 2022-06-07:
Thank you! This helped me.