360Controller/360Controller #1209: My MAC cannot see my Logitech gamepad F310

frodoe7, 2022-02-13:
### Type of Controller
All

### OS Version
Big Sur 11.4

### Driver Version
0.16.11

### Connection Method
Wired

### Device Name and Info
Logitech Gamepad F310

### Response of Controller
I tried the gamepad on windows and it working normally

but in my Mac it does not work, even after clicking the mode button , it does not open this white led (looks no power in the device)

### Response in Preference Pane
It does not see the gamepad at all (no connected devices/controllers)

toshe, 2022-04-08:
Can you check if this device is in Info.plist and if not add it device to Info.plist as a new entry and create a Pull Request ?

frodoe7, 2022-04-12:
@toshe 
It worked after following this:
https://steamcommunity.com/discussions/forum/2/3004429475624592660/

closed 2022-04-12 by frodoe7