360Controller/360Controller #1210: Xbox One Series X - "No devices found" in preferences pane

valhella, 2022-02-17:
### Type of Controller
Xbox One Series X (STANDARD GAMEPAD Vendor: 045e Product: 0b12)

### OS Version
OS Catalina Version 10.15.7

### Driver Version
Installed Version 0.16.5 because I saw in the documentation that later versions would not work with Xbox One. 

### Connection Method
Purchased USB-C to USB-A cable, I am able to see my controller identified in System Report>Hardware>USB>Controller. So no issue in getting it connected to the computer. 
My controller also successfully connects via bluetooth, my purpose in doing the wired connection is to hopefully be able to play more Steam games since it seems that many don't like the Xbox one wireless.

### Device Name and Info
>Device Name: Xbox One Series X
<br/>Product ID: 0b12
<br/>Vendor ID: 045e

### Response of Controller
Controller main light-up button works, controller vibrates when I test that using Gamepad Tester - in fact it works perfectly on Gamepad Tester (https://gamepad-tester.com/). 

### Response in Preference Pane
I see "Xbox 360 Controllers" in my system preferences. It says "No devices found". I did not have to approve the extension in my security and privacy pane, I'm assuming because I have installed 360Controller before (on an actual xbox 360 wired controller, that worked perfectly). 

### Detailed Explanation of Issue
My issue is that 360Controller just doesn't see my Xbox One controller, even though the computer does and tester sites like Gamepad Tester are able to identify it. Should I get a different version of 360Controller? Manual install? Have I screwed up by trying multiple versions of 360Controller?
My mac is ancient, late 2013, if that makes a difference. 

Thank you!
-Val

toshe, 2022-04-08:
Can you check if this device is in Info.plist and if not add it device to Info.plist as a new entry and create a Pull Request ?

Jimgitsit, 2022-04-15:
Checked mine and it is **not** in there.
Controller:

  Product ID:	**0x0b12**
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	5.09
  Serial Number:	3039373130303232383832313130
  Speed:	Up to 12 Mb/s
  Manufacturer:	Microsoft
  Location ID:	0x02100000 / 2
  Current Available (mA):	500
  Extra Operating Current (mA):	0

Are you adding these as you get them?

stevenjamescox, 2022-05-24:
Pull request #1224 adds this controller.