360Controller/360Controller #1214: Xbox One Controller is found by device but not Xbox Controller Driver

PoseidonGames, 2022-03-19:
Xbox One Controller

10.15.7

0.16.11

Wired MicroUSB

>Device Name: Controller
<br/>Product ID: 0x02dd
<br/>Vendor ID:0x045e (Microsoft Corporation)

After I plug my Xbox One controller into my mac with the micro USB cable it vibrates and shows up under the USB tab in the system report as controller. 

Nothing shows up in the controller test screen. All I see is the "No devices found" message. 

I have the latest driver installed and I have followed all the steps under the read me under [https://github.com/360Controller/360Controller#my-controller-doesnt-work](url). My issue doesn't fit under any of the other categories except for my controller doesn't work, and nothing I do with my controller affects it. Is there something with my version of macOS that changes how the controller is picked up by the driver?

Dr4oude, 2022-03-29:
Me i can't download the last alpha6 !

toshe, 2022-04-08:
Can you check if this device is in Info.plist and if not add it device to Info.plist as a new entry and create a Pull Request ?

PoseidonGames, 2022-04-11:
I'm not sure how to find that. How do I find the Info.plist files for this?

Jimgitsit, 2022-04-15:
Same problem here. @toshe, what info.plist file are you referring to?