360Controller/360Controller #1218: Merge Info.plist requests and release a new version

toshe, 2022-04-08:
@FranticRain 
I know the driver is not actively maintained, but please consider merging the open pull requests - this will add several new devices that would work perfectly fine with the current driver.

Here's a list of new controllers so far: 
#1194, #1190, #1197, #1211, #1210, #1209, #1208, #1201, #1187, #1171, #1183, #1168

Here's a list of pull requests, ready to be merged:
#1094, #1151, #1167, #1217

I tried compiling the driver but I can't seem to compile against 10.10 using the newest XCode. 
Can somebody help with the compilation? Do you need an older XCode / SDKs?

Apart from the small driver installation inconvenience this driver works perfectly well and if just by adding some entries to Info.plist we can add support for 16 new devices --- why not? :)

ValdoGhafoor, 2022-04-22:
Hey, any news with this release ?

stevenjamescox, 2022-05-24:
Out of respect for @FranticRain's time and his **years** of unpaid labor, it behooves us to do as much leg-work as possible. 
He shouldn't be asked to sort through open issues, etc., for us.

Regarding the above list:
#1187, #1183, #1168, and #1194 - already in plist
#1197 - doesn't provide IDs but older version worked, must already be in plist
#1209 - closed by original poster, the issue was related to the Steam-input kext

Submissions #1171 & #1210 are included in new pull request #1224.

I added the controllers from #1190, #1201, #1208, and #1211 and several other open issues to pull request #1223.