360Controller/360Controller #1220: Original wired Xbox360 controller not found on Monterey 12.2.1

agrebin, 2022-04-10:
### Type of Controller
 Xbox 360

### OS Version
12.2.1 Monterey

### Driver Version
0.16.11
### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
No devices found

### Detailed Explaination of Issue
Driver seems to be loaded, tried disabling and re-enabling it, but still does not found on pane, on system report controller is listed on usb with vendor and product ids, once connected the controller only flashes the central button but is not recognized by the pane.
Apr 10 10:21:18 HAL9k legacyLoader-x86_64[555]: Enable/disable driver stuff: will change state...
Apr 10 10:21:18 HAL9k legacyLoader-x86_64[555]: Will Disable Driver...
Apr 10 10:21:22 HAL9k legacyLoader-x86_64[555]: ...done!
Apr 10 10:21:23 HAL9k legacyLoader-x86_64[555]: No devices found? :( device count==0, i==0
Apr 10 10:21:32 HAL9k legacyLoader-x86_64[555]: Enable/disable driver stuff: will change state...
Apr 10 10:21:32 HAL9k legacyLoader-x86_64[555]: Will Enable Driver...
Apr 10 10:21:39 HAL9k legacyLoader-x86_64[555]: ...done!
Apr 10 10:21:40 HAL9k legacyLoader-x86_64[555]: No devices found? :( device count==0, i==0

arcbueno, 2022-05-03:
Same here, i tried with all lastest releases (stable and alpha). 
My OS Version is 12.3 (21E230) - Monterey

ianchrisman, 2022-05-12:
I also have this problem, Wired xbox360 doesn't work.  Device not found.  Controller doesn't light up but I know it works, when I dual boot into Windows it works.  Monterey 12.3.1.
I can't even delete the driver anymore, it errors out when uninstalling.

apocalypse2012, 2022-05-20:
Any status on a Monterey compatible version?

justin-tg, 2022-06-03:
I'm in the same boat, the guide button just flashes green but fails to connect (Xbox 360 Wired - Monterey 12.3.1)

TinoSM, 2022-06-13:
Check https://github.com/360Controller/360Controller/issues/1139