360Controller/360Controller #1221: Xbox Series Controller Detected by System, but not showing in panel

XeroraYT, 2022-04-10:
### Type of Controller
Xbox Series X/s Controller

### OS Version
Mac OS High Sierra - 10.3.6

### Driver Version
Every version - Clean install after each

### Connection Method
USB A to USB C

### Device Name and Info
>Device Name: Xbox Series X Controller
<br/>Product ID: 0x0b12
<br/>Vendor ID: 0x045e

### Response of Controller
Vibrates on plug in - nothing else

### Response in Preference Pane
No device found

### Detailed Explanation of Issue
Installing and fully uninstalling every version of the driver that was recommended for the One/Series controllers returns no usable result. Security & Privacy pane also **DOES NOT** display anything regarding an extension block, or any software block. 

Upon seeing previous responses from sone responding to other issues posted, regarding editing the plist file in the folder for the driver, that folder doesn't appear to be anywhere on my mac's file system upon install/restart. (hidden files **ARE** enabled.)

As per the folder not appearing, the Security & Privacy pane also doesn't display any restriction messages during the install. 

This appears to be an ongoing problem for god knows how many people, across the git, reddit, youtube, even commenters on "journalist" sites that have covered a tutorial on how to install the driver and use a controller.

**The last update was March 15, 2020. _This needs to be addressed ASAP._**

stevenjamescox, 2022-05-25:
1. This driver is completely free. And no one is entitled to @FranticRain's uncompensated labor. Nor his money—an Apple Developer account (and the certificates that can be obtained therewith) costs $99 per year.

2. This controller's information is set to be added via pull request #1224. It's possible that no update will ever be released, but all of the code is here. Thorough instructions are provided in the README regarding how one can compile this software on their own, with or without a certificate.

XeroraYT, 2022-05-26:
I forgot to update the posting with the Information I forgot to add in a
day or so later. I did ultimately build it myself and added the info needed
to recognize the Xbox Series controller, and it was usable.

Unfortunately my use case which required the controller was not met with
the results I was hoping for with system I had at the time of this posting
so I ended up scrapping the build, and sold the MacBook. (I did get it for
free so, not complaining otherwise.)

Feel free to mark this as closed/resolved.

N0rdskog, 2023-08-16:
> 

Hey there, ZeroraYT. I'm facing similar issues like you, although with an Xbox 360 controller. I'm on the same OS. Would you care to share your fork for the project? It could prove applicabl to me, and maybe even others. 

Did you by any chance try to use the controller for a Max patch? That's what I'm attempting!

XeroraYT, 2023-08-16:
Unfortunately as stated in my previous comment I scrapped the build and
sold the system I used it on. I primarily use my 5800x3d/4090 system so I
felt no need to back up my fork. Apologies.