360Controller/360Controller #1225: No driver version works with X ZHANG 360 Game Controller to XBOX 360 on mac OS Big Sur 11.6.6

aspepper, 2022-06-04:
### Type of Controller
Xbox 360 (X ZHANG works in XBOX 360 Console)

### OS Version
macOS Big Sur 11.6.6

### Driver Version
"360Controller 0.16.11 Notorized"
"360Controller 1.0.0-alpha.1"
"360Controller 1.0.0-alpha.2"
"360Controller 1.0.0-alpha.3"
"360Controller 1.0.0-alpha.4" (no Succcess on install. Because, as of this version, the installer gives error)

### Connection Method
Wired

### Device Name and Info
Device Name: X ZHANG 360 Game Controller
Product ID: X-360-3
Vendor ID: N/A 

### Response of Controller
No reaction. Just guide button blink continuously. But blinks since wire is connected on usb port.

### Response in Preference Pane
Nothing works. Game Control is not recognized.

### Detailed Explaination of Issue
After test some driver version, game controller does not works. Game Control is not recognized. And option "Enabled Driver" never keep checked.

Link to Issue Video: https://youtu.be/Ig7amluC7VI

Tks for all.
Alex Pimenta

stevenjamescox, 2022-06-05:
Alex, thank you for the report!

It appears that your video is set to private, so I cannot view it.

Are you able to provide the controller's Product ID and Vendor ID [from System Report/System Information?](https://github.com/360Controller/360Controller#how-do-i-find-my-vendor-id-and-product-id)

It seems likely that your controller isn't yet listed in [360Controller.kext's plist](https://github.com/360Controller/360Controller/blob/master/360Controller/Info.plist). I can submit them to be to the developer to be added to the driver, but the developer may have moved on from working on this driver. There might never be an update. If so, you might need to [manually add your controller to your driver](https://github.com/360Controller/360Controller#adding-third-party-controllers). You'd need to do what that ReadMe says AND one additional command in Recovery (a change since the ReadMe was updated): after `csrutil disable` then you'd need to do `csrutil enable --without kext` rather than just `csrutil enable`.

For what it's worth, I've found the alpha-5 and alpha-6 versions to be more reliable on Catalina/Big Sur/Monterey. But if the device's Product ID and Vendor ID aren't already listed in 360Controller.kext's plist, they still won't work.

Dev-ManishTomar, 2022-06-09:
You made these drivers work on Monterey How? A few months back I tried a lot but nothing worked. can you please a youtube guide for people like me, Thanks.

aspepper, 2022-06-20:
Thank you @Dev-ManishTomar, 

I just buy a new Macbook Air M1. I'm wating for usb c hub arrive at my home to do tests with Monterey. 

I'll keep informed about it, as soon come my usb c hub.

Alex Pimenta