360Controller/360Controller #1227: Could someone add Steelseries Stratus Duo? Vendor ID: 0x1038 Thank you!

johnnybegoode77, 2022-06-13:
### Type of Controller
Replace this text with: Steelseries Stratus Duo (Xbox 360)

### OS Version
10.14.6

### Driver Version
Xbox 360 Controller Driver
Version 1.0.0-alpha.2

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Steelseries Stratus Duo
<br/>Product ID: 0x1430 (Wireless)
<br/>Product ID: 0x1431 (Wired)
<br/>Vendor ID: 0x1038

### Response of Controller
N/A

### Response in Preference Pane
No, No Device Found

### Detailed Explaination of Issue
Never had a chance to get it to work.
No Device Found

stevenjamescox, 2022-06-13:
I have the same controller. It's set to be added in Pull Request #1223.

Unfortunately, the developer may have moved on from working on this driver. There might never be an update.

There _is_ a way to get this controller working for you, though. [This part](https://github.com/360Controller/360Controller#adding-third-party-controllers) of the README goes in detail about how to add controllers by adding them to the Info.plist file inside of 360Controller.kext.

Before doing that I recommend using the [Alpha-5](https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.5) build. That's the version that's worked best for my 10.14.6 machine.

Here's a [copy of the Info.plist](https://github.com/360Controller/360Controller/files/8888135/Info.plist.zip) I've been using to get my SteelSeries Stratus Duo to work on 10.14.6. The file is probably only compatible with the Alpha-5 build--and you'd need to follow the instructions in the README section I linked above regarding disabling signing requirements for any changed Info.plist to work at all.

stevenjamescox, 2022-06-13:
Excellent, you've got Alpha-5 installed.

You'll then need to [disable signing requirements](https://github.com/360Controller/360Controller#disabling-signing-requirements) as described in the README because this plist is modified from the original (having had the SteelSeries Stratus Duo wired and 2.4GHz receiver profiles added).

The installed plist to replace is located at `<name-of-your-hard-drive>/Library/Extensions/360Controller.kext/Contents/Info.plist`.
(You can right-click on 360Controller.kext and select `Show Package Contents` to view files within.

johnnybegoode77, 2022-06-13:
1st step: Failed 
Disabling signing requirements

`Jons-MacBook-Pro:~ jon$ sudo nvram boot-args="kext-dev-mode=1"
Password:
nvram: Error setting variable - 'boot-args': (iokit/common) not permitted`

```
Jons-MacBook-Pro:~ jon$ sudo kextcache -m /System/Library/Caches/com.apple.kext.caches/Startup/Extensions.mkext /System/Library/Extensions
unrecognized option /System/Library/Caches/com.apple.kext.caches/Startup/Extensions.mkext
usage: kextcache -prelinked-kernel <filename> [options] [--] [kext or directory]
       kextcache -system-prelinked-kernel
       kextcache [options] -prelinked-kernel
       kextcache -invalidate <volume> 
       kextcache -update-volume <volume> [options]
       kextcache -system-caches [options]

use kextcache -help for an explanation of each option
```

```
Jons-MacBook-Pro:~ jon$ sudo csrutil disable
csrutil: failed to modify system integrity configuration. This tool needs to be executed from the Recovery OS.
```

2nd Step: I replace the info.plist with yours
Select All - Copy, - then Select All - Paste, then Save
Is this the right way to do it?
[image]

3rd Step: Ok
`Jons-MacBook-Pro:~ jon$ sudo chown -R root:wheel /Library/Extensions/360Controller.kext `

4th Step:
```
Jons-MacBook-Pro:~ jon$ sudo kextutil /Library/Extensions/360Controller.kext
Untrusted kexts are not allowed
Kext with invalid signature (-67030) denied: /private/var/db/KernelExtensionManagement/Staging/tmp.YOT2qx/C066648C-9B03-43E1-8AF6-145CDCF3C231.kext
Bundle (/Library/Extensions/360Controller.kext) failed to validate, deleting: /private/var/db/KernelExtensionManagement/Staging/tmp.YOT2qx/C066648C-9B03-43E1-8AF6-145CDCF3C231.kext
Unable to stage kext (/Library/Extensions/360Controller.kext) to secure location.
```

5th Step: After rebooting
Still, no device was found and now I can't even Enable Driver.
[image]

stevenjamescox, 2022-06-13:
Yikes! Super un-fun to get those error messages. Thank you for providing an exceptional level of detail!

The steps `csrutil disable` and `csrutil enable --without kext` need to be run from within [Recovery Mode](https://support.apple.com/en-us/HT201314).

Terminal is found in Recovery Mode under `Utilities` in the tippy-top menu bar.

johnnybegoode77, 2022-06-14:
It works!!! Thank you so much!!! 

[image]

johnnybegoode77, 2022-06-15:
Steelseries also sent me this Engine 3.20.0:
https://steelseries.com/gg/downloads/engine/3.20.0/darwin
saying that it might detect the gamepad. 
It did not and might be because we had installed 360Controller.
I did not want to uninstall 360Controller after all the trouble I went through,
so I would never know if Engine 3.20.0 would work.
Feel free to try it out.

Suraj-Gov, 2022-07-16:
Hey @stevenjamescox I got the Stratus Duo for my M1. 

When I follow step 3d, I got this error:
```
Incompatible architecture: Binary is for x86_64, but needed arch arm64e
```

Can you let me know how to fix this issue on an Apple Silicon laptop?

stevenjamescox, 2022-07-16:
Hi @Suraj-Gov,

Insofar as I am aware, the binaries in the 360Controller releases are only compiled for x86_64 (the architecture used by Intel processors in Macs). To work, the project's binaries would need to be compiled for arm64 (the architecture used by Apple Silicon).

I have _not_ been able build this project from source (whether for x86_64 _or_ arm64). **I'm too much of a noob** right now to figure out how to properly adapt the project from its Xcode 7(?) state to a recent enough Xcode version that supports Apple Silicon on macOS. But I believe it would basically go something like this for your situation:

- Clone the full project source to your computer
- Open project in a version of Xcode capable of compiling for arm64
- Manually add the two (one for USB 2.4GHz receiver, one for wired by USB cable) SteelSeries Stratus Duo profiles to the Info.plist, following the pattern used by the many other controllers' entries
- Do Xcode & programming wizardry to get the full driver to build for arm64 _<- I'm woefully unqualified for this part_

Potential alternative that I also don't know how to do:

- Rather than rebuilding the entire driver, _only_ recompile the driver's _binaries_ from source _for arm64_. I imagine that it might be possible to do this via the command line. But, again, I'm not qualified to say that with any certainty.

@dreamingthink in the thread for [#1333 (comment)](https://github.com/360Controller/360Controller/issues/1133#issuecomment-751438560) **has partially built the project for M1,** but it looks like only some parts of 360Controller are functional therein.

As @FranticRain has explained in [#1128 (comment)](https://github.com/360Controller/360Controller/issues/1128#issuecomment-736584705), the best long-term solution would be for someone to develop a **System Extension** using DriverKit.

I wish I could be of better assistance.