360Controller/360Controller #1228: New Controller Request: PDP XBox Series X Blue Camo

PhilCoder123, 2022-07-10:
### Type of Controller
Series X

### OS Version
10.14.6

### Driver Version
Version 1.0.0-alpha.5

### Connection Method
Wired

### Device Name and Info
>Device Name: PDP Wired Controller for Xbox Series X - Blue Camo
<br/>Product ID:	0x02d8
<br/>Vendor ID:	0x0e6f

### Response of Controller
System Info can find device. Otherwise no indication of working.

### Response in Preference Pane
Device not found in preference pane.

### Detailed Explanation of Issue
I was hoping to add this controller to info.plist, but I am unable to build the kext afterwards due to the signing certificate.

stevenjamescox, 2022-07-16:
Thank you for the report! I have added your controller to Pull Request #1223.

To be clear, it's very possible that this driver will never be updated. The primary developer has moved on from this project.

But you may be able to get around the signing certificate problem using steps similar to those in thread for [Issue #1227 (comment)](https://github.com/360Controller/360Controller/issues/1227#issuecomment-1154365499) using an Info.plist you've added your controller to.

PhilCoder123, 2022-07-17:
Thank you @stevenjamescox!
I had used https://github.com/360Controller/360Controller/issues/1227#issuecomment-1154365499 as a reference when trying to get it done myself. I was able to add my controller to the info.plist the same way you did in your pull request, however I couldn't get kextutil to successfully run, even after going into recovery mode and disabling csrutil. What I did find I was that when you added your own kext into one of the comments, I had no issues running kextutil on that, but I still as not able to modify info.plist with my own controller and have the command run smoothly.
How did you compile the kext to work? Maybe I'm missing a step somewhere.

stevenjamescox, 2022-07-17:
I'm sorry that your modified kext isn't cooperating with you, @PhilCoder123!

Based on your described experience and some of my failed attempts, my hypothesis is that macOS doesn't like _already installed_ kexts being altered. The version of `360Controller.kext` I linked in the #1227 thread wasn't specially complied. I only altered its `Info.plist`. I zipped it up (via right-click "Compress" in Finder) for the sole purpose of making it postable on GitHub.

I believe this modified kext ended up working on our systems because we installed it (the full `360Controller.kext`) as one whole piece (including the `Info.plist` inside) **_after_** alteration.

Because you've already disabled kext-signing requirements _and_ modified the Info.plist, I'd recommend doing summarized steps 6-7 and 9-15. I'm including the full set of steps (and also an insanely detailed set afterwards) for others who may stumble upon this thread that need them.

### Summarized Steps:

1) Boot into Recovery Mode
2) Terminal: enter  `csrutil disable` 
3) Terminal: enter `csrutil enable --without kext`
4) Boot into normal macOS
5) Install the Alpha-5 release of 360Controller (for Mojave and later)
6) Terminal: `sudo cp -r /Library/Extensions/360Controller.kext ~/Downloads/`
7) Terminal: `sudo rm -r /Library/Extensions/360Controller.kext`
8) Using Xcode, edit the Info.plist located at `~/Downloads/360Controller.kext/Contents/Info.plist` adding new Xinput controllers according to the pattern therein — then save your changes
9) Terminal: enter `sudo cp -r ~/Downloads/360Controller.kext /Library/Extensions/`
10) Terminal: `sudo chown -R root:wheel /Library/Extensions/360Controller.kext`
11) Terminal: `sudo kextload /Library/Extensions/360Controller.kext`
12) You should now be prompted to authorize the kext in  `System Preferences -> Security & Privacy` — go do so
13) Terminal: `sudo kextcache -i /`
14) Reboot
15) Test

### Overkill Verbose Steps:

0) If you've already installed 360Controller, it might be best to [fully uninstall it](https://github.com/360Controller/360Controller/issues/1139#issue-782719045). But it might not matter. AFAIK, re-installation of any version will overwrite the old files just fine.
1) Boot into [Recovery Mode](https://support.apple.com/en-us/HT201314).
2) Open the Terminal app (Terminal is found in Recovery Mode under Utilities in the tippy-top menu bar)
3) Type `csrutil disable` then press Enter/Return (disables System Integrity Protection).
4) Type `csrutil enable --without kext` then press Enter/Return (re-enables all elements of System Integrity Protection _except_ kext signing requirements).
5) Reboot into normal macOS.
6) Download the [Alpha-5 release of 360Controller](https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.5). (I have found Alpha-5 to work well in all _full_ macOS version releases from Mojave forward. I have not tested Ventura betas.) There are four files listed on the linked page. The 1.55MB .dmg file is the one needed. (I'd direct-link the .dmg file, but seeing and using the legitimate release page is gonna feel more secure for many people.)
7) Mount the downloaded .dmg file, and open the installer therein. Install the Alpha-5 release of 360Controller. (It's okay if the package installer says there were errors at the end. This is "normal" on Big Sur and Monterey.)
8) Reboot.
9) Open the Terminal app (located at `/Applications/Utilities/Terminal.app`).
10) Type `sudo cp -r /Library/Extensions/360Controller.kext ~/Downloads/` then press Enter/Return (creates a copy of the installed `360Controller.kext` in the user Downloads folder).
11) Type `sudo rm -r /Library/Extensions/360Controller.kext` ("uninstalls" `360Controller.kext` [which is but one element of the entire 360Controller driver]).
12) In your Downloads folder at `~/Downloads/`, (which means `[your hard disk]/Users/[your user name]/Downloads/`) find `360Controller.kext`. [Right-click](https://support.apple.com/guide/mac-help/right-click-mh35853/mac) on it and select `Show Package Contents`.
13) Open the `Contents` folder.
14) Open `Info.plist`. It's an XML file. It might easiest to avoid typos by editing it with Xcode. But even simple text editors like the included TextEdit at `/Applications/TextEdit.app` can get the job done. [Atom](https://atom.io/), [Visual Studio Code](https://code.visualstudio.com/), and [Sublime Text](https://www.sublimetext.com/) are also very good.
15) Add additional Xinput controllers as desired, following the pattern found therein. If you don't want to add additional lines of code, find an existing controller entry you definitely won't need and replace its Product ID and Vendor ID with those of a controller you're trying to add. The Product ID and Vendor ID values must be in **decimal** (normal human numbers) format. In System Profiler on your Mac, your connected controller's IDs are shown in **hexadecimal** format. If you're editing Info.plist with Xcode, it will automatically convert **hexadecimal** values placed in these files to **decimal** format. If editing the plist with another program, one can use an [online converter](https://www.rapidtables.com/convert/number/hex-to-decimal.html) to get the needed **decimal** values.
16) Save your changes to the `Info.plist`.
17) Go back to the Terminal app.
18) Type `sudo cp -r ~/Downloads/360Controller.kext /Library/Extensions/` then press Enter/Return ("installs" the modified kext).
[... cut ...]
21) Go to `System Preferences -> Security & Privacy` and authorize the kext. It's okay that it says something like "Unidentified Developer." It says that because you changed the `Info.plist` and did not sign the modified kext with an active Apple Developer ID.
22) Go back to the Terminal app.
23) Type `sudo kextcache -i /` then press Enter/Return (rebuilds the system's kextcache, including your modified and now-authorized `360Controller.kext` therein).
24) Reboot.
25) Test whether your controller is detected in the Xbox 360 Controllers `System Preferences` pane.

PhilCoder123, 2022-07-17:
Hi @stevenjamescox,
Unfortunately still no luck on my end. I completely uninstalled the driver and started again with alpha 5 in case anything was lingering on. I completed the steps highlighted above and I still get errors when running `kextcache -i /`
`/Library/Extensions/360Controller.kext not in base path /System/Library/Extensions for identifier->path cache`
Any idea if I need to manually change anything in the /System/Library path? It looks like it is pointing to the kext in /Library/Extensions, but perhaps it may need an update.

stevenjamescox, 2022-07-18:
Dang! I was hoping we had it. That's quite unfortunate.

You're on Mojave 10.14.6 as indicated in your original issue report, correct?

After seeing your message, I ran `sudo kextcache -i /` on a 10.14.6 system here with Alpha-5 installed (with a modified .plist) and I get the same error message you did, _**but the driver works**_. My SteelSeries Stratus Duo is detected and responsive in the Xbox controllers preference pane. I don't believe you need to change anything in `/System/Library/Extensions/`. I believe that the alias at `/System/Library/Extensions/360Controller.kext` isn't problematic. But I could very well be wrong. What is the "Date Modified" for the alias on your system at `/System/Library/Extensions/360Controller.kext`?

I assume that you did a reboot at the end and have tried testing the controllers in the Xbox controllers preferences pane?

I'm baffled. I don't yet have any suggestions. I will, though, try installing a modified version of Alpha-5 on another Mojave system tonight. 

For what it's worth, here are the relevant lines I get when I run `sudo kextcache -i /` on the aforementioned 10.14.6 system:
```
username$ sudo kextcache -i/
Kext with invalid signatured (-67030) allowed: <OSKext 0x7fe20bd246e0 [0x7fffa2557920]> { URL = "file:///Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
Invalid signature -67030 for kext <OSKext 0x7fe20bd246e0 [0x7fffa2557920]> { URL = "file:///Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
KernelCache ID: 4A75938FFABB0E0EFF6017024358A995
Kext with invalid signatured (-67030) allowed: <OSKext 0x7f8fb00d0ee0 [0x7fffa2557920]> { URL = "file:///Library/Extensions/360Controller.kext/", ID = "com.mice.driver.Xbox360Controller" }
/Library/Extensions/360Controller.kext not in base path /System/Library/Extensions for identifier->path cache.
```

PhilCoder123, 2022-07-18:
My output looks about the same as yours, I just get a few more messages and some different hex values so probably no worries there. I tested my controller on a windows system and it gets detected there, just to rule out any potential controller issues. I don't get detected on the preference pane, nor does the LED go on when plugged in so I can try some further debug.
The date on the system extension path does seem to be about a week old, despite me having deleted the file and re-copied it over today. I'll read up on if I can do something to relink or refresh it. Thanks again for your time!