360Controller/360Controller #1230: "No Device Found" with Xbox One controller (non Bluetooth versionI on M1, Monterey

Sogrea, 2022-08-12:
### Type of Controller
Xbox One

### OS Version
12.4, Macbook Pro M1 (2020)

### Driver Version
1.0.0 alpha 2, 3, 4, 5, 6

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller
<br/>Product ID: 0x02d1
<br/>Vendor ID: 0x045e (Microsoft Corporation)

### Response of Controller
No response from the controller

### Response in Preference Pane
Nothing shows up in the preference pane, only the "No Devices Found" option.

### Detailed Explaination of Issue
I actually tried all the alpha versions, some of them install successfully, until alpha 3, then from alpha 4 the installation doesn't finalize, but at least after a reboot it appears to be installed.
I even tried rebuilding the project for arm64 from xCode, but unfortunately it didnt' work.

JamesL007, 2022-08-16:
I have the same issue as well on Monterey.

achak98, 2022-08-20:
I have the same issue as well. Someone, please help.

antonkuskov, 2022-10-21:
same issue with xbox 360 controller + m1 chip + monterey

marcustut, 2023-12-19:
Mine was working perfectly on Ventura and after formatting it to Sonoma then it stopped working

zwlong6, 2024-12-15:
> Sogrea

Could you give me your driver?