360Controller/360Controller #1232: XBox 360 Controller: "No Device Found" - usb wired controller - new device?

morpe, 2022-08-20:
### Type of Controller
Xbox 360, usb wired

### OS Version
Catalina 10.15.7

### Driver Version
360Controller 0.16.11

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Zexrow - Controller per Xbox 360
<br/>Product ID: 0x028e
<br/>Vendor ID:  0x045e  (Microsoft Corporation)

### Response of Controller
the controller blinks for a while, then turns off.
works with GeForce Now on OSX

### Response in Preference Pane
Preference Pane: No devices found

### Detailed Explaination of Issue
---

ehargy, 2024-02-02:
I am having this same issue and I have the same product and vendor ID!! 

Did you solve the problem??

SNinjaBeard, 2024-07-29:
I'm also having this issue with the same product ID and vendor ID, OS version is 14.5.