360Controller/360Controller #1234: Installation fails for alpha.6 and 0.16.8 not working

LucasKovacs, 2022-08-21:
### Type of Controller
Xbox One Wireless Controller

### OS Version
macOS Monterey 12.5.1

### Driver Version
I tried with multiple including alpha 6 and 360Controller 0.16.8 which seemed to be the latest with Bluetooth support.

### Connection Method
Bluetooth

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x02FD
<br/>Vendor ID: 0x045E

### Response of Controller
It works perfectly with GeForce now.

### Response in Preference Pane
On System preferences it says that no devices are found.

### Detailed Explaination of Issue
With 0.16.8: It gets connected through bluetooth, GeForce now and Steam Big View are working, but the controller is not recognised under system preferences. It does nothing, even with legacy mode enabled.

With alpha.6 the installation fails and cannot continue. I'm never asked to allow access or something, it just fails saying installation failed.

PhlipPhlops, 2022-09-03:
identical issue here

jetmartin, 2022-09-10:
Same here, "installation failed".
Using an usb controller.
Device found on USB system report, but "no device found" on System Preferences > Xbox 360 controllers tab.

leifnagtegaal, 2022-09-10:
unfortunately also for me: "Installation failed

jetmartin, 2022-09-11:
Hi,
It's Fixed on my side :
I've redo the whole procedure here : https://github.com/360Controller/360Controller/issues/1139
The install error persist as described.
Had (like the first time) some issues on the _$ sudo kmutil load -p /Library/Extensions/360Controller.kext_ cmd but finally works and require a reboot.
I can't say if it's du to a _vi_ of the file or other commands I've tryed.
Bellow the comands I've done :

- spctl kext-consent status
- spctl kext-consent list
- vi /Library/Extensions/360Controller.kext
- sudo kmutil load -p /Library/Extensions/360Controller.kext

Hope this helps.

uTnOJkji5quPSNE5, 2022-09-22:
Hi guys,
On macOS 12.6 (uname -an: Darwin Kernel Version 21.6.0: Mon Aug 22 20:17:10 PDT 2022; root:xnu-8020.140.49~2/RELEASE_X86_64 x86_64) still failed to load the kext after the "vi."

$ sudo kmutil load -p /Library/Extensions/360Controller.kext
Error Domain=KMErrorDomain Code=71 "Unsupported Error: one or more extensions are unsupported to load 	Kext com.mice.driver.Xbox360Controller v1.0 in executable kext bundle com.mice.driver.Xbox360Controller at /Library/StagedExtensions/Library/Extensions/360Controller.kext
Unsupported Error: one or more extensions are unsupported to load 	Kext com.mice.driver.Xbox360Controller v1.0 in executable kext bundle com.mice.driver.Xbox360Controller at /Library/StagedExtensions/Library/Extensions/360Controller.kext" UserInfo={NSLocalizedDescription=Unsupported Error: one or more extensions are unsupported to load 	Kext com.mice.driver.Xbox360Controller v1.0 in executable kext bundle com.mice.driver.Xbox360Controller at /Library/StagedExtensions/Library/Extensions/360Controller.kext
Unsupported Error: one or more extensions are unsupported to load 	Kext com.mice.driver.Xbox360Controller v1.0 in executable kext bundle com.mice.driver.Xbox360Controller at /Library/StagedExtensions/Library/Extensions/360Controller.kext}

michaelgpearce, 2022-11-06:
> Hi guys, On macOS 12.6 (uname -an: Darwin Kernel Version 21.6.0: Mon Aug 22 20:17:10 PDT 2022; root:xnu-8020.140.49~2/RELEASE_X86_64 x86_64) still failed to load the kext after the "vi."
>
> $ sudo [...]

Same error here.