360Controller/360Controller #1239: The controller is detected on MACOS Ventura, but does not work in games.

mauriciolamego, 2022-10-26:
### Type of Controller
ORIGINAL XBOX360 WIRELESS (USB SENSOR)

### OS Version
MACOS VENTURA

### Driver Version
1.5 ALPHA

### Connection Method
WIRELESS

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID: 0x0291
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Opening the console in system settings the controller responds normally, but it does not show the steady LED on the controller and when I enter any game it does not respond.

### Response in Preference Pane
the program appears in the preference panel, it even responds to commands, but it doesn't work when I enter any game. The control also does not show the connected LED, despite responding in the panel preference.

### Detailed Explaination of Issue
he program appears in the preference panel, it even responds to commands, but it doesn't work when I enter any game. The control also does not show the connected LED, despite responding in the panel preference.
I've entered safe mode, authorized the extension, but nothing solved.

aadictive, 2022-10-30:
Hey, this is happening to me as well. I just updated today

macmeister1967, 2022-11-04:
Did you both try re-syncing the controller with the button on the dongle, and the top of the controller?

mauriciolamego, 2022-11-04:
Yes, I have tried all those ways mentioned. I also tried starting in safe mode, installing previous versions of the fix and other procedures that I researched on the internet.

MAURICIO LAMEGO

> Em 4 de nov. de 2022, à(s) 06:23, macmeister1967 ***@***.***> escreveu:
> 
> Did you both try re-syncing the controller with the button on the dongle, and the top of the controller?
> 
>

bagong, 2022-11-04:
For me the install broke in Ventura. I couldn't even run uninstall. Had to manually remove the files...

macmeister1967, 2022-11-05:
There is a youtube video where someone gets errors when trying to
uninstall, so reinstalls ALPHA2 as a last-ditch try, and it worked
perfectly for him.

I understand he wasn't using Ventura, but the "cannot uninstall" errors he
hot sounds very familiar. Worth a try?

bagong, 2022-11-06:
Thanks @macmeister1967 , I'd try if there was a significant chance that reinstalling works. I think I removed the kexts cleanly, fortunately there is a script in the preference pane folder that shows what to remove. But I'd only want to go through the steps again if there was a reason why it might work this time. Is the alpha-2 Version different in a way that makes it more likely to work?

macmeister1967, 2022-11-06:
Just quoting what happened on that YouTube video... he couldn't uninstall
the drivers, but reinstalled Alpha2 and they worked first restart.

bagong, 2022-11-06:
@macmeister1967 yes, thank you.

pedskr, 2022-11-18:
@macmeister1967 I tried this but it did't work. I can't uninstall the drive and the drive dosen't work. I don't know what more to do...

PatiB87, 2023-08-06:
As far as I know, the issue is not this driver but macOS Ventura.
I also use it on an M2 Max and tested an Xbox One Bluetooth gamepad which is officially supported. Therefore it connects fine, yet at least every 2nd app does not recognize it (the OFFICIALLY supported controller!).

See for instance here:
https://www.reddit.com/r/macgaming/comments/xmcd51/updating_to_macos_ventura_beta_8_may_break/?rdt=65063