360Controller/360Controller #1240: Xbox 360 Wired Controller is not detected by driver on MacOS Ventura

DanyaONELOVE, 2022-11-07:
### Type of Controller
Xbox 360 (I'm not sure if that's an original one though) 

### OS Version
Ventura 13.0 (22A380)

### Driver Version
Alpha 6; Alpha 2; 0.16.11;

### Connection Method
Wired (USB -> TypeC adapter. The adapter is working.)

### Device Name and Info
  Device Name: N/A
  Product ID:	0x028e
  Vendor ID:	0x045e  (Microsoft Corporation)

### Response of Controller
The guide button light up. All 4 parts of it. It just flashes once every half a second.

### Response in Preference Pane
The controller does NOT appear in the preference pane. 

### Detailed Explaination of Issue
Steam don't see the device, and it doesn't work in the games as well. The driver says 'No devices found'.

ymgenesis, 2023-02-05:
Same issue. Wired 360 controller, Monterey 12.3. No matter which version I install, no devices are found in the preference pane.

cory-eheart-1, 2023-03-18:
Same issue. Cannot use wired controller on Monterey

lwlxy, 2023-03-24:
Same issue here

clousky2020, 2023-05-06:
same issue ,wait

AjayDextrous, 2023-08-06:
is there any way to solve this?

zellwk, 2023-08-25:
I got the same issue as well. Ventura 13.2.