360Controller/360Controller #1244: PARDON ME, BUT

jimmijam10, 2022-11-24:
Not only is this completely useless, is the XBOX controller able to be connected WIRELESSLY through BLUETOOTH with ease, CONTRASTING this "BUSINESS INSIDER" article, BUT IT SEEMINGLY does NOTHING AT ALL.   >:(   Is this a VIRUS!?

bagong, 2022-11-26:
No, it's not, it has been helping people for years to connect cabled X360 gamepads to the Mac, including cheap clones. So expect it made 1000s of kids happy. Unfortunately it's not maintained any more and some hackery is required to make it work on newer MacOS releases. I am not sure it works on Ventura at all. Would be lovely to have it updated...

Therises, 2022-11-27:
@bagong well, it's even doesn't work on Monterey (M1 Max).

monokrome, 2023-01-10:
@Therises @jimmijam10 This is because the signed version of the package is expired due to having not been updated since May 2020. The package is probably also not developed for M1 macs since they were not released prior to the latest 360 controller drivers release.