360Controller/360Controller #1245: Need Driver for OSX13.0.1

jc470gp, 2022-11-29:
### Type of Controller
Replace this text with: Original Xbox, Xbox 360, Xbox One, All or N/A

### OS Version
Replace this text with the version of macOS that you are running

### Driver Version
Replate this text with the version or versions of the driver that you are using or have tested

### Connection Method
Replace this text with: Wired, Wireless Adapter, Bluetooth, N/A or more specific method if applicable

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name:
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Replace this text with information about the controller's reaction in the preference pane. Does the controller appear in the preference pane? Do all of the buttons, joysticks, and triggers act as expected?

### Detailed Explaination of Issue
Replace this text with a detailed explaination of your issue. Simply saying "it doesn't work" is a quick way to ensure that your issue is never fixed. Without detailed information on your specific issue, we cannot fix your issue. Including as much information as possible in this section increases the likelyhood of getting your issue quickly and effectively resolved.

gabrielucido, 2022-12-03:
Waiting for it =)

jc470gp, 2022-12-08:
> Waiting for it =)

Ok, I know it’s on voluntary base. Please take a time but it is not so long hopefully. Thank you.

gabrielucido, 2023-01-09:
I'm a web developer, can I help with programming? I know C#, JS, and Python and have a MacBook 13.1

jc470gp, 2023-01-10:
Thank you for the response. I want to use 360gaming controller on OSX 13, but the driver is not exist anywhere . If you are able to develop and release that, please share to as free software. Thank you.

jonasw8, 2023-03-30:
> I'm a web developer, can I help with programming? I know C#, JS, and Python and have a MacBook 13.1

Teria que reescrever o driver com o novo "IOKIT" para joystick. Big sur em diante requer isso e só alguns joysticks funcionam nativamente, como os de nintendo switch, xbox serie x, e alguns de playstation. A vantagem seria usar o ventura, pois nele aparece uma opção para joystick nas preferencias do sistema quando o joystick é compatível.

alguns links para vc ter uma noção

https://developer.apple.com/library/archive/documentation/ServicesDiscovery/Conceptual/GameControllerPG/Introduction/Introduction.html#//apple_ref/doc/uid/TP40013276

https://developer.apple.com/documentation/xcode/configuring-game-controllers

https://developer.apple.com/documentation/gamecontroller

Caso vc conseguir reescrever esse driver para esse novo método da apple, seu driver será único para todos os sistemas de big sur em diante