360Controller/360Controller #1249: Wired Controller shows in System Report but doesn't work with this driver macOS 12.6.2

kylejericson, 2023-01-02:
### Type of Controller
Xbox Series X EnWired

### OS Version
12.6.2
### Driver Version
Version 1.0.0-alpha.6

### Connection Method
Wired

### Device Name and Info
Xbox Series X EnWired Pastel Purple Camo:

Product ID:
Ox202c

Vendor ID:
Ox20d6

Version:
1.01

Serial Number:
00000131CD51097D

Speed:
Up to 12 Mb/s
Manufacturer:

PowerA
Location ID:
Ox14200000 / 20

Current Available (mA):
500

Extra Operating Current (mA):
0
### Response of Controller
Nothing

### Response in Preference Pane
Nothing

### Detailed Explanation of Issue
I installed the driver enabled the extensions and rebooted. Connected controller and won't show in the system prefs but will show in System Report in macOS

why520zsy, 2023-01-22:
Have you solved your problem? I also encountered this problem.

kylejericson, 2023-01-22:
Nope still can’t use this on Mac.

bzuruce, 2023-02-20:
Big Sur + is not going to work

shanbalanthetincanman, 2023-02-21:
Does anyone know if it would work on Wine?