360Controller/360Controller #1251: Shows in Settings but error on Ventura 13.3

raymondj-pace, 2023-03-31:
### Type of Controller
Xbox 360 wired

### OS Version
13.3

### Driver Version
Version 1.0.0-alpha.4

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Microsoft XBOX 360 controller wired
<br/>Product ID:
<br/>Vendor ID:

### Response of Controller
N/A

### Response in Preference Pane
No controller found

### Detailed Explaination of Issue
Cannot load driver
The driver has failed to load due to an unknown error, code: -603947007. Please report this error to the driver maintainer.
On the "Enable Driver" checkbox in settings.

clousky2020, 2023-05-06:
It's been blinking since it's been connected

Infixo, 2023-07-26:
I have the same issue. Using alpha5 version.