360Controller/360Controller #1256: How can I uninstall the Xbox 360 Controllers?

Picstory, 2023-06-13:
### Type of Controller
Xbox One

### OS Version
Ventura 13.4

### Driver Version
1.0.0-alpha.5

### Connection Method
Wired, Bluetooth

### Device Name and Info

### Response of Controller

### Response in Preference Pane
Neither wired nor bluetooth discovers the controller.

### Detailed Explaination of Issue
I tried to delete the driver but I get an error and cannot delete the driver.
How can I uninstall the Xbox 360 Controllers entry in the System Settings menu?

Yuuki-Moon, 2023-07-30:
Also having the same problem, uninstalling the drivergives me:

"Error!

Error uninstalling the driver"

without any further information

arundaniel-kennedy, 2023-08-22:
This helped actually #1139