360Controller/360Controller #1262: Trouble installing

JurajkoDS, 2023-07-10:
### Type of Controller
Xbox 360

### OS Version
macOS Ventura 13.2, MBP 2019 Intel chip

### Driver Version
None yet

### Connection Method
Wired

### Device Name and Info
N/A

### Response of Controller
None yet

### Response in Preference Pane
None yet

### Detailed Explaination of Issue
I cannot install the alpha 6 driver. Is it because its meant for M1?

lioilsources, 2023-07-14:
M2 13.4.1 (22F82) the same issue.

jackofall111, 2023-08-06:
Same issue :( macos Ventura 13.5

supersynthesis, 2023-08-08:
same issue here on Ventura 13.4.1. I tried changing the permissions of my drive to "reduced" as suggested by Apple, but it didn't fix it.

coolcat836, 2023-08-16:
i’m having the same issue/error message on Ventura 13.4.1 with the M1 chip. the suggestions in the documentation for adjusting the security settings did not work/don't apply to this version of macOS

pterodactyl-soup, 2023-08-21:
Same venture 13.2 intel macbook air.