360Controller/360Controller #1264: Kindly add support for my Claw Shoot

kayasdevastra, 2023-08-03:
### Type of Controller
Claw Shoot 

### OS Version
Ventura 13.5
### Driver Version
1.10
### Connection Method
Dongle
### Device Name and Info
>Device Name: Claw Shoot Bluetooth Controller
<br/>Product ID: 0x028e
<br/>Vendor ID: 0x045e

### Response of Controller
Everything Setup, Connected to the Macbook M1 13 inch 2020, shows up in the SYSTEM INFORMATION > HARDWARE > USB 3.1 > Xbox360 For Windows:

  Product ID:	0x028e
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.10
  Serial Number:	10F36D6
  Speed:	Up to 12 Mb/s
  Manufacturer:	ShanWan
  Location ID:	0x00100000 / 1
  Current Available (mA):	500
  Extra Operating Current (mA):	0

### Response in Preference Pane
Device Doesn't occur in the  XBOX 360 MENU.

### Detailed Explanation of Issue

Works in the steam link app but not in steams, works in gampad tester on https://hardwaretester.com/gamepad . and parallels as well as YAAGL (Yet Another Anime Game Launcher) [Genshin Impact for Windows].

In 360Controller > Info.plist My Devices Product Id and Vendor Id is not present. I tried to do fix the driver according to the Instructions given for adding 3rd party controllers. But I didn't understood many things and it's not working out for me. I'm new to macbook. Kindly help me.
It's working on parallels too

yashwant-creator, 2023-08-08:
I have the same controller as well. I have been trying to find a fix for it for SOO long!! please find a fix

arundaniel-kennedy, 2023-08-22:
@yashwant-creator and @kayasdevastra, you guys can try gamepad-tool from sdl that allows sdl2 controllers to work in sdl2 based games. I used it and was able to play it in ps3 mode but I'm sure it will work in xbox mode for you guys.

yashwant-creator, 2024-05-14:
@arundaniel-kennedy, I want to play AAA Games, not sdl2. Is there a way?