360Controller/360Controller #1267: I tried to port this to DriverKit

noah-nuebling, 2023-09-02:
Hey guys, I tried to port this driver to DriverKit, so it works under modern macOS. 

Here's the repo for my port: https://github.com/noah-nuebling/mac-gamepad-driver

Unfortunately it doesn't quite work so far but I thought I'd share it in case somebody else wants to keep working on it.

## Discussion

It creates two clients for the gamepad in the IORegistry. `gamepad_driver_user` and `Xbox360ControllerClass`. Here's a screenshot.
[image]

Here's more details on how it works:

I have an XboxOne wired controller. When I attach it to the computer, macOS creates a registry entry of class `IOUSBHostDevice` for it (it does this anyways even if this driver is not installed). Then, if this driver is installed, the system creates the `gamepad_driver_user` registry entry which is driven by the code in the cpp class `gamepad_driver_user` and attaches it to the `IOUSBHostDevice` as a client. When  `gamepad_driver_user` starts, it starts finding USB interfaces and stuff of the device and it tries to determine what kind of gamepad is being used (XboxOne, Xbox360, XboxOriginal, and some other options).  It tries to find the USB endpoints on the device for sending and receiving data. Then it writes some data to the device. It asks the system to create another registry entry which is driven by the code in the cpp class `Xbox360ControllerClass` and the system attaches this new entry to `gamepad_driver_user` as a client. Then `gamepad_driver_user` starts reading from the device in a loop. The data it reads from the device in a loop is sent to `Xbox360ControllerClass→handleReport()`, which then does some additional processing on the report before passing it on to userspace clients. 

The `Xbox360ControllerClass` registry entry is a subclass of `IOHIDDevice`. Which means userspace programs can find it and start talking to it using the `IOHIDManager` apis.  This architecture is the same as the original 360Controller driver. But the way it's implemented is pretty different because all the APIs changed a lot for DriverKit. 

My current implementation doesn't work properly for my wired XboxOne controller because all of the reports that are being read from the device don't seem to be "valid" according to the logic I copied from the 360Controller driver. I don't know if my controller is weird or if there's a bug in my code.

Also see `Notes.md` in the repo for more info and steps to run and test the driver.

Another thing that's really annoying is that you need to request permissions from Apple to attach your driver directly to the `IOUSBHostDevice`. And you need to request separate permissions for each vendorID. This might make it infeasible to ever publish the DriverKit based driver. But currently I'm using Ad Hoc signing and some other hacks to get around this requirement and run the driver even without permission from Apple. I even requested the permissions from Apple but they declined :(  

Another side note: 

Here are the logs that the driver outputs on my system:
[image]

I think next steps for further developing the driver would be: 
- Testing and debugging with different controllers
- Implementing the proper support for XboxOne and XboxOriginal controllers (stubs for XboxOne controller support are already in the repo)
- Implementing support for peripherals like the microphone if anyone wants to.
- Maybe building a UI / copying the UI over from the original 360Controller driver. (Although I would recommend porting it from being a prefpane to being a normal app, because prefpanes are really hard to develop under modern macOS.)
- Maybe building remapping and deadzone customization features like the original 360Controller driver

If you have anymore questions, let me know!

Coresi7, 2023-09-14:
And you, my friend, are a hero.

noah-nuebling, 2023-11-08:
@jonasw8 Nope, unfortunately I won't spend time on this any time soon, because I want to focus on my project Mac Mouse Fix. 

I'm not sure much will come from this even if somebody spends a lot of time since it seems that you need permission from Apple for each controller you want to support and they denied it to me for the controller I have.

Hopefully the community will figure something out, but it's not looking great :/

kailous, 2024-01-26:
Your effort to port the 360Controller to DriverKit is seriously awesome – you're a hero for taking this on.

Have you thought about reaching out to third-party game controller manufacturers? Although the macOS gaming market isn't as vast as Windows, they might still be interested in supporting macOS. Your project could catch their attention, perhaps opening up opportunities for support or collaboration.

Keep it up! I'm really looking forward to the progress of your project; my third-party controller has been collecting dust for far too long.

jonasw8, 2024-09-23:
Hi Guys! Breaking News

Sequoia received support for new joysticks

-     Wired Xbox 360
-     Nintendo Switch N64 Controller
-     Nintendo JoyCon Grip
-     Wired Xbox ONE

Would it be possible to add support for wireless xbox360?

kokroo, 2024-09-26:
@noah-nuebling You might want to join us at https://github.com/WiiController/VirtualController/issues/1

We are continuing the goal of your project in a way, and your experience with Apple's entitlement process and your insight into the inner workings of Xbox 360 controllers will be useful and very much appreciated!

noah-nuebling, 2024-09-27:
@kokroo thank you for your invitation! I don't think I have much to offer in this space aside from the information you can find in this thread and in the [the repo](https://github.com/noah-nuebling/mac-gamepad-driver). However, if you have any specific questions that you think I might be able to help you with feel free to reach out!