360Controller/360Controller #1273: Sonoma....

RainEgypt, 2023-11-18:
### Type of Controller
Xbox 360

### OS Version
Sonoma 14.1.1. and previous Sonoma...., M2 Pro processor

### Driver Version
0.16.11

### Connection Method
Wireless Adapter

### Device Name and Info
>Device Name: Some Chinese "PC Wireless Gaming Receiver"
<br/>Product ID: 0x02a9
<br/>Vendor ID:0x045e  (Microsoft Corporation)
Version: 6.0a

### Response of Controller
Just blinking after push connection button

### Response in Preference Pane
I used this devices at Asus laptop on Windows. All was OK, but now laptop broken. Now I see Xbox 360 Controller in system preferences and can enable driver, but nothing happen. I means no connection, no devices found.

### Detailed Explaination of Issue
I followed by instruction here: https://github.com/360Controller/360Controller/issues/1139
MANUALLY UNINSTALLING THE DRIVER - OK
Update your Controller Firmware - I did it when used devices with old Asus laptop.
Step by step, 1-7 - OK
but at last step 8, when I try to execute command in terminal I received this:
Error Domain=KMErrorDomain Code=71 "Incompatible architecture: Binary is for x86_64, but needed arch arm64e
Unsupported Error: one or more extensions are unsupported to load: 	Kext com.mice.driver.Xbox360Controller v0.16.11 in executable kext bundle com.mice.driver.Xbox360Controller at /Library/Extensions/360Controller.kext" UserInfo={NSLocalizedDescription=Incompatible architecture: Binary is for x86_64, but needed arch arm64e
Unsupported Error: one or more extensions are unsupported to load: 	Kext com.mice.driver.Xbox360Controller v0.16.11 in executable kext bundle com.mice.driver.Xbox360Controller at /Library/Extensions/360Controller.kext}

Maybe I need to make something additional?

luizgununes, 2023-12-05:
Same here, still looking for some solution. If I knew what is the problem exactly I could look at the code, but seems to be something with latest security updates from Apple.

pinguinjkeke, 2023-12-28:
https://github.com/360Controller/360Controller/issues/1139#issuecomment-1374882029

This solution works, but I was able to load the driver only with `csrutil disable`