360Controller/360Controller #1279: Xbox One Controller not found by driver

Heyl0, 2024-03-13:
### Type of Controller
Xbox One

### OS Version
13.2.1 (22D68)

### Driver Version
0.16.11_Notarized

### Connection Method
Wired

### Device Name and Info
(If you don't know this information, please refer to the README for how to find it. Enter N/A if not applicable)
>Device Name: Controller
<br/>Product ID: 0x02ea
<br/>Vendor ID: 0x045e  (Microsoft Corporation)

### Response of Controller
Replace this text with information on how the physical controller reacts. Does the guide button light up? Does it vibrate? Etc.

### Response in Preference Pane
Driver "No devices found"
### Detailed Explaination of Issue
Xbox symbol glows
USB Device registers in Hardware Information
Inputs not interpreted

Heyl0, 2024-05-08:
1. MacBook Air 2019
2.  Yes

Florian-Martinez, 2024-05-08:
That's odd. Did you allow the driver in Privacy and Security in the settings app?

Heyl0, 2024-05-13:
yes I did that's why I am really confused why the driver isn't recognising the device that is connected

Florian-Martinez, 2024-05-14:
Check [this thread](https://github.com/360Controller/360Controller/issues/1139) its for MX, but theres all the steps explained.