360Controller/360Controller #1281: Not at all allowing me to install

tynkerbell2, 2024-04-18:
### Type of Controller
Xbox 360

### OS Version
Sonoma 14.4.1 

### Driver Version
[360Controller 1.0.0-alpha.6](https://github.com/360Controller/360Controller/releases/tag/v1.0.0-alpha.6)

### Connection Method
Bluetooth

### Device Name and Info
N/A

### Response of Controller
N/A

### Response in Preference Pane
N/A 

### Detailed Explaination of Issue
I does not work AT ALL.

I tried installing this, but my mac wont allow me to do it. 
Screenshots below:
[image]

Florian-Martinez, 2024-04-27:
Bluetooth support has been disabled in a previous version. This driver only allows to support wired method. You need to disable the SIP as well before installing the driver.  (Recovery > Terminal > type csrutil disable)

Purplepie, 2024-04-27:
Thank you @Florian-Martinez, 
I try to plug  a **Xbox One controller (model 1708)** wired in micro-USB (with the right data cable) via a Thunderbolt adapter.
I followed your steps, disabled the SIP in Terminal on Recovery mode, but it still doesn't work for me on my **MacBook Pro M1 — MacOS: Sonoma 14.4.1 (23E224) —**
It displays the exact same "_The installation Failed_".

Although the installation failed, the "**Xbox 360 Controller"** app is present in **Settings**, but displays "No devices found". Tried to plug/unplug the controller, Restarting the Mac, nothing better :(

Florian-Martinez, 2024-04-28:
It does not add up, do you use the latest version from there https://github.com/360Controller/360Controller/issues/1139#issuecomment-2025825668

enrmini, 2024-06-30:
> Bluetooth support has been disabled in a previous version due to a high risk of panic issues. This driver only allows to support wired method. You need to disable the SIP as well before installing the driver. (Recovery > Terminal > type csrutil disable)

I have a Xbox 360 wired controller and Mac 14.5 gives me the same problem.

enrmini, 2024-06-30:
> First of all uninstall your driver and do it again. It does not add up, do you use the latest version from there [#1139 (comment)](https://github.com/360Controller/360Controller/issues/1139#issuecomment-2025825668) or build it yourself and follow the steps from the thread i gave you. It should works as i have the same version as you.

At Recovery mode, I do not have acess to the Terminal in Mac 14.5.

DrKoneko, 2024-08-31:
> At Recovery mode, I do not have acess to the Terminal in Mac 14.5.

Search up how to disable System Integrity Protection on mac and it'll show how.