360Controller/360Controller #1288: Installation fail on MacOs 15

zk-731, 2024-11-09:
On MacOs 15, it will fail to install with unknown issue but it will appear in settings. I saw some tutorial saying that we need to enable the drivers, by enabling it, the whole setting will stuck forever, also not able to detect my controller.

NoX1De, 2024-12-23:
+1 on this, the installer appears to be broken and fails the installation on MacOS Sequoia 15.1

[image]

Checking "Enable Driver" in MacOS settings just totally freezes up the settings panel and soft-crashes it, you have to manually close and reopen the settings panel to get the "Xbox 360 Controllers" add-on section to display again.  This is very broken and needs attention to get this working on Sequoia.

@MaddTheSane @Drewbadour Would appreciate it if you guys may be able to find some time to take a look, thank you!  Most older console Emulators do not work with Xbox One controllers on MacOS so this driver is important for MacOS Emulator users.

MaddTheSane, 2024-12-23:
As of 15.1, there is a native wired driver pre-installed with a driver for the wired 360 controller. Support for OSes beyond Big Sur would require a re-write of the driver. With the native GameController driver in 15.1, this third-party driver wouldn't be needed if you just needed controller (and rumble) support.

NoX1De, 2024-12-23:
Thanks, interesting I'll have to test that out...but I want to point out the problem also exists where this is just stuck in the MacOS settings panel because it's so broken, the Uninstall button does not work to remove it and throws an error...not great, I'll see if I can find a way to remove it manually but this seems pretty stuck atm without more digging.

iFloneUEFN, 2024-12-25:
You need to build it for arm64e.

iFloneUEFN, 2024-12-25:
I managed earlier to make it run by doing this on 15.2 but its only showing the inputs in the settings, not in the browser and not working either on iOS apps. You would ask me why would you do when theres already support for xbox controller, well because i did not find a way to change the controller deadzone for iOS app on Mac. Do you have any solution in mind @MaddTheSane?